collin80/SavvyCAN #770: Issue with SavvyCan(V213) slowdown during frame transmission in fuzzing

ksmagala, 2024-04-22:
Hi,

I would like to address performance issues encountered while sending frames using fuzzing within the ID range of 0x000 to 0x7FF, encompassing a total of 2048 different identifiers. I noticed that after sending the first 1000 frames, the program begins to slow down significantly. With a set interval of 10 ms, the program should send about 100 frames per second, but the rate drops to merely 10-20 frames. Performance only returns to normal after exceeding the 2048 frame mark.

Interestingly, a similar issue occurs when the ID range is set from 0x000 to 0x3FF (1024), with the slowdown resolving after surpassing 1024 received frames.

For testing, I am using the ESP32RET module and SavvyCan software version V213. It is worth noting that with the previous software version, V199, these problems did not occur. I am confident that the issue is not hardware-related, as I am using a computer equipped with an i7-11850H processor, 32GB of RAM, a Samsung 990 PRO SSD, and running Windows 11.

Has anyone else experienced similar issues or could suggest possible solutions?

Thank you in advance for any assistance.

collin80, 2024-04-23:
That sounds suspiciously like a bug in SavvyCAN but I'm not sure where/why right now. It's weird that it would happen like you're describing but your description is quite helpful for me to track this down. Are you sequentially scanning the address space in each instance?

ksmagala, 2024-04-23:
I conducted a frame transmission speed test on a 2020 13” MacBook (i5, 16GB, OS Ventura). After initiating the test on MacOS, I noticed that within the ID range from 0x000 to 0x7FF, the frame transmission speed slows down. When the count exceeds 2048, the frame transmission rate per second does not increase as it does with the program running on Windows 10, and remains within the range of 5-30 frames per second. Below, I am also attaching screenshots.
[image]
[image]