collin80/SavvyCAN #778: SavvyCAN “Save” crashes Mac OS 10.13.6

TessieCatEC, 2024-05-09:
Hi Collin,  I am trying to save a SavvyCAN frame but each time I select “Save” (using any of the 3 save buttons) SavvyCan crashes the application back to the desktop.  I have over 13GB available space and I am only trying to save a few minutes of data from MrDIY device CAN shield with an ESP32 and is loaded with an open source project called ESP32RET.  

Everything seems to work fine with Wi-Fi connection (GVRET).  Data flows in… After just a few minutes I “Suspend Capturing” and then use Menu->File->Save Decoded Frames.  Instantly, “poof”, app has crashed back to desktop.  An Error window displays to send to Apple.  It’s the same result when I select: “DBC File Manager” and “Save Decoded Frames CVS”.

Am I doing something wrong?  Thanks so much for your help.  Steve@teamameritest.com.   614-562-7979 txt/cell

collin80, 2024-05-13:
It appears that the issue is that dylibs are being used that expect a MacOS version of 10.15 or higher. And, since you're using 10.13 this is not satisfied and it dies. Here is the actual crash reason:

Dyld Error Message:
Symbol not found: ____chkstk_darwin
Referenced from: /Users/USER/Desktop/SavvyCAN.app/Contents/MacOS/../Frameworks/
libpcre2-16.0.dylib (which was built for Mac OS X 10.15)

So, if you can upgrade to 10.15 then it might work. However, I am noticing that you're using an older MacBook Air and have only 4GB of RAM so maybe 10.15 won't work. I use OpenCore Legacy Patcher on my MacBook Pro 2013 to run MacOS 14.3 even though it is not supposed to run anything newer than MacOS 10.15. However, you might be able to use OpenCore to get up to at least 10.15 so that it works. Though, I suppose I can't promise that it'll work even then but I know it does work on MacOS 14.

TessieCatEC, 2024-05-25:
Thank you!  Yes, I purchased a newer MacBook Air OSX 12.6.1 and the items in the file menu don’t crash the app.  HOWEVER, the newer Mac is not allowing me to select “CAN-Wireless” in the network menu.  A DIALOG window is requiring me to enter a WAP-2 password.
I can’t find anywhere on GitHub or SavvyCAN site that shows a password.   
Do you have any idea how I can solve this issue? 
On my older Mac, I don’t recall needing a password.

Thanks so much for any help you offer.
Steve

TessieCatEC, 2024-05-25:
Thanks for your help.  I purchased a newer MacBook Air with OSX 12.6.1…. All the File -> Menu options now work as anticipated.   Thanks again.