collin80/SavvyCAN #781: Timestamps on Windows are not human-readable

SebKuzminsky, 2024-05-13:
# Good on Linux

On a local Linux build of commit 67dfaf1, the timestamps look very clear and understandable:
[image]

# Bad on WIndows
However, on the Windows build (exe downloaded from the github CI build of that commit) the timestamps look "raw", no matter if `Normalize Frame Timing` is clicked or not:
[image]
[image]

collin80, 2024-05-13:
In the preferences, see what the time stamp format is set to. I'll bet it is set differently for each of your operating systems. The Windows time stamps look like microseconds. If you set it to seconds that would look a little better. If you set it to system clock instead then it'll look like what the Linux version looks like.

SebKuzminsky, 2024-05-14:
Ah, thanks, I'll take a look and get back to you!

SebKuzminsky, 2024-05-14:
You're right, fixing it in Preferences worked.  So this is operator error on my part.  Thanks!

closed 2024-05-14 by SebKuzminsky