collin80/SavvyCAN #787: Crash after selecting "Enable Continuous Logging" when no connection active

mdear, 2024-05-23:
I invoke SavvyCAN on my Linux Mint device, see that no frames are coming in and no connections exist, select "Enable Continuous Logging" and the app crashes.

```
vcuadmin@vcuproject2030:~/Downloads$ ./SavvyCAN-53c0dae-x86_64.AppImage 
Gtk-Message: 07:27:03.737: Failed to load module "xapp-gtk3-module"
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 1714, resource id: 8666035, major code: 40 (TranslateCoords), minor code: 0
QMetaObject::invokeMethod: No such method FrameSenderObject::stopPlayback()
vcuadmin@vcuproject2030:~/Downloads$ ./SavvyCAN-53c0dae-x86_64.AppImage 
Gtk-Message: 07:28:10.108: Failed to load module "xapp-gtk3-module"

(SavvyCAN-53c0dae-x86_64.AppImage:55650): Gtk-WARNING **: 07:28:15.205: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Mint-Y/mimetypes/16/image.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Mint-Y/mimetypes/16/image.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Aborted (core dumped)
vcuadmin@vcuproject2030:~/Downloads$ 
```

collin80, 2024-05-23:
Do you, by chance, use Wayland instead of X11? I know various people have reported problems like this where it complains about Gtk related things and I thought it was due to the use of Wayland.

phatpaul, 2024-10-01:
This looks like the same crash as reported in here:
#725 
#621 
#414

The fix suggested by [@TadayukiOkada ](https://github.com/collin80/SavvyCAN/issues/621#issuecomment-1637605255) fixes the issue.  The steps are pretty straightforward:

```sh

# Open a terminal and cd to the directory containing the AppImage.
# Extract the AppImage:

./SavvyCan-x86_64.AppImage --appimage-extract
# (This creates a directory called squashfs-root)

# removing libpango from the AppImage fixes the problem:
rm ./squashfs-root/usr/lib/libpango-1.0.so.0*

# Download the AppImage creation tool and make it executable:
wget https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage
chmod +x appimagetool-x86_64.AppImage

# Make a new AppImage using the updated directory:
./appimagetool-x86_64.AppImage squashfs-root

# This will create an updated AppImage named SavvyCan-x86_64.AppImage

```