collin80/SavvyCAN #789: mac build artifact does not include QT Serial Bus support

SebKuzminsky, 2024-05-25:
To reproduce:
* Run the Mac executable from the github CI
* Open the Connection manager window
* Select Add New Device Connection
Note that the QT SerialBus Devices option is grayed out and not available.

collin80, 2024-05-29:
Yes, this is true. I know it's happening but I'm yet unsure why. It certainly should be included but it seems that the MacOS packaging tools aren't doing so. I haven't figured out how to install the required files yet. I thought someone had done it but I don't remember who figured it out. They certainly didn't fix the CI build if they had figured it out on their end. So, it's still on my list of things I need to do.

theforcedk, 2024-05-31:
The libraries are included, and loaded.. however I can't figure out why it's grayed out. I don't see any particular errors.. perhaps a debug build would help, as there is very little output from the executable itself.

**Print loaded dylibs**
```
#:~ DYLD_PRINT_TO_FILE=~/savvy_out /Volumes/SavvyCAN/SavvyCAN.app/Contents/MacOS/SavvyCAN

#:~ grep -i 'qtserial' ~/savvy_out 
dyld[31470]: <4302337E-CBE9-3760-9A7F-13E523FAFB0E> /Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialBus.framework/Versions/5/QtSerialBus
dyld[31470]: <06BAC677-ECD8-3710-8B9D-F2156E51E5A3> /Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialPort.framework/Versions/5/QtSerialPort
dyld[31470]: dyld_image_path_containing_address(0x10b7a3000) => '/Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialBus.framework/Versions/5/QtSerialBus'
dyld[31470]: dyld_image_path_containing_address(0x10b71b000) => '/Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialPort.framework/Versions/5/QtSerialPort'```
```

**Look for libraries in the mounted (DMG) archive**
```
#:/Volumes/SavvyCAN ls /Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/*/ | grep -i qtserial
/Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialBus.framework/:
QtSerialBus
/Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialPort.framework/:
QtSerialPort
```

**Verify QT version for libraries (5.15.0+)**
```
#:/Volumes/SavvyCAN otool -L /Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/*/qtserial*
/Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialBus.framework/QtSerialBus:
	@executable_path/../Frameworks/QtSerialBus.framework/Versions/5/QtSerialBus (compatibility version 5.15.0, current version 5.15.13)
	@executable_path/../Frameworks/QtCore.framework/Versions/5/QtCore (compatibility version 5.15.0, current version 5.15.13)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	@executable_path/../Frameworks/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.15.0, current version 5.15.13)
	@executable_path/../Frameworks/QtSerialPort.framework/Versions/5/QtSerialPort (compatibility version 5.15.0, current version 5.15.13)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1300.36.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
/Volumes/SavvyCAN/SavvyCAN.app/Contents/Frameworks/QtSerialPort.framework/QtSerialPort:
	@executable_path/../Frameworks/QtSerialPort.framework/Versions/5/QtSerialPort (compatibility version 5.15.0, current version 5.15.13)
	@executable_path/../Frameworks/QtCore.framework/Versions/5/QtCore (compatibility version 5.15.0, current version 5.15.13)
	/System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1953.255.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1300.36.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1319.0.0)
```

**Application output**
```
#:/Volumes/SavvyCAN lldb /Volumes/SavvyCAN/SavvyCAN.app/Contents/MacOS/SavvyCAN
(lldb) target create "/Volumes/SavvyCAN/SavvyCAN.app/Contents/MacOS/SavvyCAN"
Current executable set to '/Volumes/SavvyCAN/SavvyCAN.app/Contents/MacOS/SavvyCAN' (x86_64).
(lldb) r
Process 40076 launched: '/Volumes/SavvyCAN/SavvyCAN.app/Contents/MacOS/SavvyCAN' (x86_64)
2024-05-31 21:48:27.417432+0200 SavvyCAN[40076:16508607] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=40076
2024-05-31 21:48:27.417583+0200 SavvyCAN[40076:16508607] SecTaskCopyDebugDescription: SavvyCAN[40076]/0#-1 LF=0
2024-05-31 21:48:28.479831+0200 SavvyCAN[40076:16508607] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=40076
2024-05-31 21:48:28.479953+0200 SavvyCAN[40076:16508607] SecTaskCopyDebugDescription: SavvyCAN[40076]/0#-1 LF=0
2024-05-31 21:48:28.583303+0200 SavvyCAN[40076:16508607] QFSFileEngine::open: No file name specified
2024-05-31 21:48:28.583335+0200 SavvyCAN[40076:16508607] QFSFileEngine::open: No file name specified
2024-05-31 21:48:28.583344+0200 SavvyCAN[40076:16508607] QFSFileEngine::open: No file name specified
2024-05-31 21:48:28.583352+0200 SavvyCAN[40076:16508607] QFSFileEngine::open: No file name specified
2024-05-31 21:48:28.646322+0200 SavvyCAN[40076:16508607] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.
2024-05-31 21:48:32.731960+0200 SavvyCAN[40076:16508607] [miscellany] CLIENT ERROR: TUINSRemoteViewController does not override -viewServiceDidTerminateWithError: and thus cannot react to catastrophic errors beyond logging them
2024-05-31 21:48:33.653066+0200 SavvyCAN[40076:16508756] [StateRestoration] BOOL _NSPersistentUIDeleteItemAtFileURL(NSURL *const __strong) Failed to stat item: file:///Users/snip/Library/Saved%20Application%20State/com.yourcompany.SavvyCAN.savedState/restorecount.plist
[... cut ...]
2024-05-31 21:48:41.370633+0200 SavvyCAN[40076:16508607] SecTaskCopyDebugDescription: SavvyCAN[40076]/0#-1 LF=0
2024-05-31 21:48:41.433268+0200 SavvyCAN[40076:16508607] [miscellany] CLIENT ERROR: TUINSRemoteViewController does not override -viewServiceDidTerminateWithError: and thus cannot react to catastrophic errors beyond logging them
2024-05-31 21:48:43.796480+0200 SavvyCAN[40076:16508607] QMetaObject::invokeMethod: No such method FrameSenderObject::stopPlayback()
Process 40076 exited with status = 0 (0x00000000) 
```

simoneruffini, 2024-06-05:
Any updates on this?