collin80/SavvyCAN #806: SavvyCAN with Qt 6.7.2 on Linux: no Qt Serial port for 'passthroughcan' device

douglasheld2, 2024-06-25:
I prepared Qt 6.7.2 release and then built SavvyCAN against it on ElementaryOS 7.1 on a mid-2010 MacbookPro (Intel x86_64) per steps outlined in https://github.com/collin80/SavvyCAN/discussions/804

I have a no-name "Mini-VCI" j2534 OBD2 pass-through cable. When I plug in this cable it registers as a serial device on /dev/ttyUSB0. (note, it does not load as a socket_can interface like /dev/can0)

Steps to reproduce:

1. Plug in USB passthrough cable.
2. `journalctl -f` confirms the device is hooked up to /dev/ttyUSB0
3. start SavvyCAN and open the "Connection Settings" window
4. select "enable console" and select "Add New Device Connection"
5. In the "New Connection" window, select "QT SerialBus Devices...
6. Under SerialBus Device Type, choose "passthroughcan"
10. The "Port" combo box remains blank so nothing can be selected
11. However, "/dev/ttyUSB0" can be typed into the combo box for the port setting.
12. Click "Create New Connection"
13. A red LED on the cable PCB changes, presumably indicating a connection state change
8. on the SavvyCAN stdout, it's printed "qt.canbus: Can not disconnect an unconnected device."
9. The "Connectied Devices" table updates with a new device of type "SerialBus", subtype "passthrucan", port /dev/ttyUSB0, Buses=1, and Status changes from "Disconnected" to "Connected".
10. Packets are not collected from the CAN bus.

Expected result:
1. Collect CAN frames
2. Or, an error message indicating the device is not supported or not responding as expected
3. True status as to whether the device is "connected" or not

Actual result:
It is unclear what "connected" may mean, and whether this device is truly connected. It could be that a good serial connection is established to the device's PCB - but for some reason the CAN packets are not transmitted.
It could be that this device can not theoretically work in the way I am expecting it to. If that is the case some clarity would be nice.

Remarks:

1. I'm happy to try this again and collect more specific debugging information. Please provide instructions.
2. I'm happy to start experimenting with older versions of Qt, and older releases of SavvyCAN until I find something that works - but as that may be time consuming, I would gladly take any hints on how far back to go.

collin80, 2024-06-25:
Passthrough has always been a bit of a pain. For one, I think it normally requires a pass through driver. These tend to only exist for Windows. Then, some person who shall remain nameless seems to have locked the passthrough support to 500k so if you need a different speed it also tends not to work. Truth be told, it has been years since I last tried to use a passthrough driver device. Other people have successfully done it but I think they did need to edit the speed because it at least used to be hard coded in the source code. 

QT has some online documentation about the implementation of passthrough: https://doc.qt.io/qt-6/qtserialbus-passthrucan-overview.html

douglasheld2, 2024-06-26:
Thank you very much for the information. I am dumbfounded that the J2534 driver is actually part of the Qt framework... and that they set out to support it with so many limitations.

Arinerron, 2025-06-10:
Sure am glad it's a userspace driver on Linux, haha