collin80/SavvyCAN #808: Timestamp is missing in recorded trace after loading

janaranjanahl, 2024-06-27:
I have logged CAN trace and while live traffic I can plot and see the time stamps as seen below trace.
[image]

When I load the trace to SavvyCAN, it loses the timestamp info as shown below and can not see the data in a plot. I wonder I am doing something wrong and is there a fix for this?

[image]

janaranjanahl, 2024-06-27:
Update:

I can open the following trace created from PCAN wothout an issue.
[image]

in SavvyCAN
[image]

janaranjanahl, 2024-06-27:
I noticed selecting normalize frame timing will do the trick. But it is not clear this button is checked or not. I suggest we need to keep this a Preferences setting.

santhanraghava, 2025-04-29:
I am also facing the same issue of loosing timestamps when trying to load a .trc file which was logged by SavvyCAN. What are other options to solve if I can't log data again in PCAN.