collin80/SavvyCAN #810: Graph Settings / Add This Graph modal dialog too tall

douglasheld2, 2024-07-02:
The minimum system requirements include screen resolution of 1024x768. The "Add a New Graph" window however, doesn't fit on such a screen.

Steps to reproduce

1. Install SavvyCAN on a MacbookPro7,1 which has screen resolution 1280x800.
2. Select RE Tools -> New Graphing Window
3. Right-click in the graph area
4. Select "Add new graph"

Expected results
Fill out the form and click "Add This Graph" button

Actual results
The bottom of the window falls off the bottom of the usable screen

Proposed results
Rearrange the controls so the whole window fits within 1024x768
Or, add a vertical scroll bar?
Or failing that, adjust the system requirements to the smallest usable screen size

collin80, 2024-07-10:
Yes, it is an issue. Originally all the screens did fit on a 1024x768 screen but there has been some drift over time. It probably doesn't help that my main machine is 2560x1600 so none of the screens come anywhere close to falling off the end on my screen. I have a MacBook Pro (2013 model) but it also has a greater than 1080p screen resolution. So, I suppose my problem is that all my screens are huge resolution and so I got a bit lax in making sure that things still fit smaller resolutions. It may be possible to rearrange the controls to make things fit once again.

collin80, 2024-07-10:
I have rearranged the new graph window a bit so that it fits on an 800 pixel tall screen. This is a good thing to do not only for you but because a number of Raspberry Pi screens aren't any larger and I do remember having this issue on a Raspberry Pi. The new layout should be a little better (haven't tested on the RPI though)

closed 2024-07-10 by collin80