collin80/SavvyCAN #820: How to get Lilygo TCAN485 working on Savvycan

nagup, 2024-07-27:
So I saw a youtube video where a guy is trying to use a car battery as storage for solar energy. He is using the Lilygo485 board on savvycan and I am wondering how he did that. I also want to get my board to work on savvycan, the board comes blank its just a multipurpose board, not with candlelight or the other (protocols?) names you see here like LAWICEL. 

Any help appreciated thanks

DocOnGitHub, 2024-11-28:
Maybe try contacting the guy in the video, or at least share enough information for someone else to find the video and look into it.

MotorvateDIY, 2024-12-13:
Use the recently updated version of A0RET (https://github.com/collin80/A0RET)
In the config.h set the CAN TX/RX pins for the board:

// Set CAN TX/RX pins
#define GPIO_CAN_TX GPIO_NUM_27
#define GPIO_CAN_RX GPIO_NUM_26

Also, somewhere in setup() set GPIO23 (aka CAN_SE_PIN) pin low, to enable the SN64HVD231 "high speed" mode:
  pinMode(23, OUTPUT);
  digitalWrite(23, LOW);

Compile and flash the project. 
Connect the board to power/USB and a CAN bus.

On the PC, connect to the "ESP32SSID" wifi.
In SavvyCAN, click on Connection, Open Connection Window.
Click on Add New Device Connection, Network connection (GVRET), type in the IP address of:
192.168.4.1 and Create new connection.

You now have a CAN bus to Wifi connection feeding data into SavvyCAN.
This is how I use SavvyCAN 99% of the time. It is fast and no need for cables in the car or on the desk.

Good Luck!

Board Docs here: https://github.com/Xinyuan-LilyGO/T-CAN485

demon1300, 2025-02-20:
> Use the recently updated version of A0RET (https://github.com/collin80/A0RET) In the config.h set the CAN TX/RX pins for the board:
>
> // Set CAN TX/RX pins #define GPIO_CAN_TX GPIO_NUM_27 #define [...]

I can't get this to compile.  Then again, i'm no developer...  Any ideas why this would be happening?

It's complaining about CAN1 not being defined in a bunch of the files but from what i can tell, it's all there in the right places:

`C:\Users\*username*\Desktop\A0RET\lawicel.cpp: In member function 'void LAWICELHandler::handleLongCmd(char*)':
C:\Users\*username*\Desktop\A0RET\lawicel.cpp:169:17: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  169 |                 CAN1.sendFrame(outFrame);
      |                 ^~~~
      |                 CAN0
C:\Users\*username*\Desktop\A0RET\lawicel.cpp:203:50: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  203 |                 if (!strcasecmp(tokens[4], "X")) CAN1.setRXFilter(0, filt, mask, true);
      |                                                  ^~~~
      |                                                  CAN0
C:\Users\*username*\Desktop\A0RET\lawicel.cpp:204:26: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  204 |                     else CAN1.setRXFilter(0, filt, mask, false);
      |                          ^~~~
      |                          CAN0
C:\Users\*username*\Desktop\A0RET\lawicel.cpp:232:17: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  232 |                 CAN1.begin(speed, 255);
      |                 ^~~~
      |                 CAN0
C:\Users\*username*\Desktop\A0RET\gvret_comm.cpp: In member function 'void GVRET_Comm_Handler::processIncomingByte(uint8_t)':
C:\Users\*username*\Desktop\A0RET\gvret_comm.cpp:221:57: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  221 |                 if (out_bus == 1) canManager.sendFrame(&CAN1, build_out_frame);
      |                                                         ^~~~
      |                                                         CAN0
C:\Users\*username*\Desktop\A0RET\gvret_comm.cpp:345:21: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  345 |                     CAN1.begin(settings.CAN0Speed, 255);
      |                     ^~~~
      |                     CAN0
C:\Users\*username*\Desktop\A0RET\gvret_comm.cpp:350:22: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  350 |                 else CAN1.disable();
      |                      ^~~~
      |                      CAN0
C:\Users\*username*\Desktop\A0RET\SerialConsole.cpp: In member function 'void SerialConsole::handleConfigCmd()':
C:\Users\*username*\Desktop\A0RET\SerialConsole.cpp:249:13: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  249 |             CAN1.begin(settings.CAN0Speed, 255);
      |             ^~~~
      |             CAN0
C:\Users\*username*\Desktop\A0RET\SerialConsole.cpp:252:14: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  252 |         else CAN1.disable();
      |              ^~~~
      |              CAN0
C:\Users\*username*\Desktop\A0RET\SerialConsole.cpp:258:40: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  258 |             if (settings.CAN1_Enabled) CAN1.begin(settings.CAN1Speed, 255);
      |                                        ^~~~
      |                                        CAN0
C:\Users\*username*\Desktop\A0RET\SerialConsole.cpp:266:17: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  266 |                 CAN1.setListenOnlyMode(true);
      |                 ^~~~
      |                 CAN0
C:\Users\*username*\Desktop\A0RET\SerialConsole.cpp:268:17: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  268 |                 CAN1.setListenOnlyMode(false);
      |                 ^~~~
      |                 CAN0
C:\Users\*username*\Desktop\A0RET\SerialConsole.cpp:307:23: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  307 |         handleCANSend(CAN1, newString);
      |                       ^~~~
      |                       CAN0
C:\Users\*username*\Desktop\A0RET\can_manager.cpp: In member function 'void CANManager::setup()':
C:\Users\*username*\Desktop\A0RET\can_manager.cpp:40:9: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
   40 |         CAN1.enable();
      |         ^~~~
      |         CAN0
C:\Users\*username*\Desktop\A0RET\can_manager.cpp:56:9: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
   56 |         CAN1.disable();
      |         ^~~~
      |         CAN0
C:\Users\*username*\Desktop\A0RET\can_manager.cpp: In member function 'void CANManager::sendFrame(CAN_COMMON*, CAN_FRAME&)':
C:\Users\*username*\Desktop\A0RET\can_manager.cpp:84:17: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
   84 |     if (bus == &CAN1) whichBus = 1;
      |                 ^~~~
      |                 CAN0
C:\Users\*username*\Desktop\A0RET\can_manager.cpp: In member function 'void CANManager::loop()':
C:\Users\*username*\Desktop\A0RET\can_manager.cpp:137:16: error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?
  137 |         while (CAN1.available() > 0 && (maxLength < (WIFI_BUFF_SIZE - 80)))
      |                ^~~~
      |                CAN0
C:\Users\*username*\Desktop\A0RET\A0RET.ino: In function 'void setup()':
C:\Users\*username*\Desktop\A0RET\A0RET.ino:151:13: error: 'CAN_SE_PIN' was not declared in this scope
  151 |     pinMode(CAN_SE_PIN, OUTPUT);  // Initialize CAN_SE_PIN (GPIO23) as output
      |             ^~~~~~~~~~

exit status 1

Compilation error: 'CAN1' was not declared in this scope; did you mean 'CAN0'?`