collin80/SavvyCAN #823: Log Format Add Request

ba05, 2024-07-31:
Is it possible to read logs in this format? If not what would be involved in adding this functionality?

```
2011-04-25-10:16:43.004,2,127250,3,255,8,ff,8b,0a,ff,7f,ff,7f,fd
2020-08-22T13:53:01.536Z,7,65313,17,255,8,13,99,ff,7f,ff,ff,ff,ff
2011-04-25-10:15:37.690,6,59392,3,12,8,01,ff,ff,ff,ff,34,ff,00
2021-01-27T23:34:19.539Z,3,129029,1,255,43,ff,de,48,48,af,94,32,70,eb,69,fd,aa,6d,bc,01,00,d7,1b,22,44,ef,47,f6,88,82,ac,fe,ff,ff,ff,ff,24,fc,1b,32,00,46,00,80,f8,ff,ff,00
2020-08-22T13:53:02.193Z,2,129025,127,255,8,b0,dd,b1,1f,1b,2f,3d,03
2020-08-22T13:52:57.591Z,7,126993,6,255,8,60,ea,24,ff,ff,ff,ff,ff
```

[Link to log from canboat GitHub project](https://github.com/canboat/canboat/blob/021b1edfa98da3a799a022a95411632508eb589f/samples/sample-single.raw.txt)

collin80, 2024-07-31:
It's probably possible. But, it sure would help if I knew what all the fields mean. At first I thought it was
TimeStamp, Length, ID, byte0, byte1, etc but it's not.
The second number certainly isn't length like I thought. I'll have to dig into their code a bit to figure out what each field is.

Also, quite a lot of the traffic listed is well over 8 bytes. Some lines might even be over 64 bytes. SavvyCAN is capable of loading CAN-FD frames where there are up to 64 bytes. The file you linked to isn't raw CAN traffic but rather NMEA traffic already collected into messages from the individual frames. It might not be that easy to import.

phatpaul, 2024-12-20:
2nd the request for decoding NMEA2000 Fast Packets.

http://en.wikipedia.org/wiki/NMEA_2000

"The Fast Packet protocol defined in NMEA 2000 provides a means to stream up to 223 bytes of data, with the advantage that each frame retains the parameter group identity and priority. The first frame transmitted uses 2 bytes to identify sequential Fast Packet parameter groups and sequential frames within a single parameter group transmission. The first byte contains a sequence counter to distinguish consecutive transmission of the same parameter groups and a frame counter set to frame zero. The second byte in the first frame identifies the total size of the parameter group to follow. Successive frames use just single data byte for the sequence counter and the frame counter. Because many of the NMEA 2000 parameter groups exceed 8 bytes but do not require the 1,785-byte capacity of multi-packet, the default method of transmitting multi-frame parameter groups in NMEA 2000 is using the Fast Packet protocol.
Regardless of which protocol is used, multi-frame parameter groups are sent on a frame-by-frame basis and may be interspersed with other higher priority parameter groups using either protocol, or even single- frame parameter groups. Each device is responsible for reassembling the parameter group once all the frames for the parameter group are transmitted."