collin80/SavvyCAN #827: J1939 DBC data graphing issue

Mikelike314159, 2024-08-11:
Hello, hopefully a simple question, but I didn't find any info online.  I logged J1939 data from a heavy-duty US-based vehicle using a CSS Electronics CL2000, and I have loaded and interpreted the data in SavvyCAN with the SAE J1939 DBC I purchased from CSS. I'm now trying to graph some parameters, and am following the instructions to add parameters to a data graphing window, but no graph is appearing (and no data is present when I save the graph data to csv).  I cannot determine what I'm doing wrong.  The parameters I've selected are presented in the data, and I'm using the DBC decoding through the software.  I'm attaching a screenshot.  Can anyone help me understand what I'm doing incorrectly which is preventing my data from being graphed?  Many thanks in advance.
[image]

collin80, 2024-08-13:
When you loaded the DBC file, did you tell it that it needs to use J1939 decoding? There are three options now: Exact, GMLAN, and J1939. If you don't set it to J1939 then it won't decode the ID properly and won't work.

Mikelike314159, 2024-08-13:
Thank you for your response.  Yes, unfortunately, it is set to J1939.
In fact, it decodes in the data table (and exports as a CSV or decoded text
file correctly).  It seems to be associated only with the
plotting/graphing.  When I select the graphing option to show data points,
it does show a point at zero/zero, and when I export the plot data, it's a
blank file (with headers). It appears to be associated with decoding the
CAN data only for the graph.
I appreciate your help,
Mike

lansalot, 2024-10-23:
Came here looking to find what I was doing wrong as well - funnily enough, I was plotting exactly the same data value.

[image]

[image]

Empty graph.. Strange indeed!

---

So if I enter it manually, same thing - empty. If I try to map only a single byte manually  however, it works fine.

[image]

[image]

---

Also, if I enter something manually instead of from DBC.. (here, first byte, not much variation but should show in a graph)
[image]

Up it comes:
[image]

If I double-click on "try" label to edit it, savvycan crashes. Same on any manually-entered value - if it came from DBC, it edits fine.

Jun 13 2023 build here.

collin80, 2024-10-25:
Yup, just ran into this myself. Here's the basic problem and also why I think it'll be a little more involved to solve it fully:

J1939 obviously has information encoded into the 29 bit CAN id. The DBC files tend to be made such that they know this and leave areas that a wildcarded as 00's. For instance, I made a DBC file where there is a 0x18FFD500 message defined. That's because the last two digits in hex are the ID of the source device and could be anything. The problem here is that if you then load the details of the signal, it loads 0x18FFD500 as the ID but this won't tend to match your actual message. It doesn't do the J1939 wildcarding in the graphing window. So, you have to copy the signal parameters and then change the 00 to the relevant value of the ID you really want to graph.

Now, here's where it gets slightly more complicated. I can't simply do J1939 ID decoding because you really could (and I in fact DO have) multiple devices on the bus that send a particular message and differ only in source ID. If it wildcarded you'd get all the device's data jumbled up in a big mess. You want the graph to be only one device. So, probably what it should do is go scan for all the IDs of devices that send that given message and give you the option of which you want. But, it currently does not do that.

So, for now, you have to copy parameters and then manually fill in the missing bits of the ID to get a fully formed ID to graph. Then it does work.

MatinF, 2025-01-07:
This is an issue related to CAN ID decoding (at the full 29 bit level) vs. J1939 PGN decoding (at the 18 bit PGN level). We explain this in detail in [our J1939 intro](https://www.csselectronics.com/pages/j1939-explained-simple-intro-tutorial#decode-j1939).

You will need to match the 29-bit ID exactly when using SavvyCAN, which will not likely happen by default when you load the ID from the DBC. Therefore you should manually review what the full 29 bit ID is based on your trace view and copy/paste that to replace the default inserted DBC based ID.