collin80/SavvyCAN #834: Timestamp is always 0

FliFro, 2024-08-27:
The timestamp in the main window shows always '0', or, depending on settings, the corresponding System Time Jan-01 etc.
When I use CANgaroo to capture CAN messages, the timestamps are OK.

The problem did not occur on first installation, but after i played around with the settings. I was unable to get it back to working.
Even deleting the entire SavvyCAN folder and re-downloading the release didn't help. (Since the programm doesn't install in the classic way - where does it store the settings?)

V213, 
Windows 11
SLCAN devices (CANable 2.0)

[image]

collin80, 2024-08-28:
There is a configuration file saved. On Windows it should be in your user directory under
AppData/Local/SavvyCAN I think.

This is a reference to the standard file locations for QT apps:
https://doc.qt.io/qt-5/qstandardpaths.html

FliFro, 2024-08-28:
Thanks for the quick reply!
Found the .ini file at: AppData\Roaming\EVTV\SavvyCAN.ini, renamed it and started from scratch. 
Unfortunately, this didn't solve the issue - timestamps are still 0. 

For now, I can do what I need to do with python and cantools, but you might want to look into that one anyways. 

Keep up the good work!

Puneeth-kmp, 2024-09-07:
[image]
this will solve ur problem Recording with Time Stamp	File → Preference→  Time Keeping → Select system clock → back  → Use Start Continuous logging   → Check for Red LOGGING Text Blinking on main window . if u follow dees steps wen the logging it self u'll be able to see time stamp in log file csv or txt

threepotMR2, 2025-01-28:
I have exactly the same problem, system time stamp worked the first time I selected it.

Running windows 7, Canable 2.0 standard firmware.

Non of the time stamps work, I've deleted the ini file and started again but still nothing.

System clock in the logger window just says Jan-01 00:00:00.000

And if I pick seconds, micro,mill its just 0.0000 etc

FliFro, 2025-02-04:
> [image] this will solve ur problem Recording with Time Stamp File → Preference→ Time Keeping → Select system clock → back → Use Start Continuous logging → Check for Red LOGGING Text Blinking on main [...]

Unfortunately, that didn't help. Timestamps are always zero in log files, too.

threepotMR2, 2025-02-06:
I installed Linux version with same CAN usb hardware, and then time stamps work! So it has to be a problem with the windos executable.

wucke13, 2025-03-25:
I'm on NixOS 24.11 using savvycan 213. I tried all four settings under "Time Keeping", but none of them yielded a timestamp other than 0. Also there is https://github.com/collin80/SavvyCAN/issues/878 which seems to have the same problem.

anegrin, 2025-03-26:
`Use absolute time in decoded CSV output` is not present in the windows UI, maybe you wanna try editing `.ini` and set

```
CSVAbsTime=true
```

because of https://github.com/collin80/SavvyCAN/blob/226628582d4440ba6bb4d1633634c4f454dd573d/mainwindow.cpp#L1401

anegrin, 2025-03-27:
can anyone give this build a try:

https://github.com/anegrin/SavvyCAN/releases

it's windows only, this is the PR https://github.com/anegrin/SavvyCAN/pull/1

if it work a will open it on collin80/SavvyCAN

anegrin, 2025-03-29:
this PR https://github.com/anegrin/SavvyCAN/pull/1 kinda works: time is now progressing even if starting form 1970 (not clear why)

build for windows: https://github.com/anegrin/SavvyCAN/actions/runs/14147302974/artifacts/2844521081
build for linux: https://github.com/anegrin/SavvyCAN/actions/runs/14147302974/artifacts/2844517386

you can check them in this action: https://github.com/anegrin/SavvyCAN/actions/runs/14147302974 (while osx is still pending at the time of writing this)

builds will expire in a couple of weeks

opening PR to main repo...

anegrin, 2025-03-29:
FIXING PR https://github.com/collin80/SavvyCAN/pull/909

anegrin, 2025-05-06:
PR has been merged https://github.com/collin80/SavvyCAN/pull/909