collin80/SavvyCAN #838: graphs are not working for values  above 255 and below 0

Puneeth-kmp, 2024-08-31:
graphs are not working for values  above 255 and below 0 in the frame sender window how to fix the issue..? . problem with unsigned char in the interpretation of data. i am very new to this stuff i dont even know this the proper way of asking for help...
[image]

collin80, 2024-09-05:
Ahh, ok, yes, the graphs are all using unsigned values but you want to see the signed value. This should probably be an option somehow.

Puneeth-kmp, 2024-09-05:
[image]
as of now i did some modification to the code and now graphs are similar in view compared to how they look on matlab or jupyter notebook which were plotted using savedecoded to text file opp. but i am facing the scaling issue i am trying fix the same but stuck btw  ill share the code i think it has something to do with the fundamentals of data types. need some guidance .                               
        void FrameInfoWindow::updateDetailsWindow(QString newID)
{
    int targettedID;
    int minLen, maxLen, thisLen;
    int64_t avgInterval;
    int64_t minInterval;
    int64_t maxInterval;
    QVector<int> minData, maxData;
    QVector<QVector<int>> dataHistogram;
    QVector<double> byteGraphX;
    QVector<QVector<float>> byteGraphY; // Changed to float
    QHash<QString, QHash<QString, int>> signalInstances;
    QVector<uint8_t> changedBits, referenceBits;

    QTreeWidgetItem *baseNode, *dataBase, *histBase, *tempItem;

    if (modelFrames->count() == 0) return;

    targettedID = static_cast<int>(Utility::ParseStringToNum(newID));

    qDebug() << "Started update details window with id " << targettedID;

    avgInterval = 0;

    if (targettedID > -1)
    {
        frameCache.clear();
        for (int i = 0; i < modelFrames->count(); i++)
        {
            CANFrame thisFrame = modelFrames->at(i);
            if (thisFrame.frameId() == static_cast<uint32_t>(targettedID)) frameCache.append(thisFrame);
        }

        if (frameCache.count() == 0) return;

        const unsigned char *data = reinterpret_cast<const unsigned char *>(frameCache.at(0).payload().constData());
        int dataLen = frameCache.at(0).payload().length();

        minData.resize(dataLen, 256);
        maxData.resize(dataLen, -1);
        dataHistogram.resize(65536, QVector<int>(dataLen, 0)); // Increased to handle up to 16-bit values
        changedBits.resize(dataLen, 0);
        referenceBits.resize(dataLen);

        ui->treeDetails->clear();

        baseNode = new QTreeWidgetItem();
        baseNode->setText(0, QString("ID: ") + newID );
        
        //removed need paste back
        
        
        data = reinterpret_cast<const unsigned char *>(frameCache.at(0).payload().constData());
        dataLen = frameCache.at(0).payload().length();

        for (int c = 0; c < dataLen; c++)
        {
            changedBits[c] = 0;
            referenceBits[c] = data[c];
        }

        DBC_MESSAGE *msg = dbcHandler->findMessageForFilter(targettedID, nullptr);

        for (int j = 0; j < frameCache.count(); j++)
        {
            data = reinterpret_cast<const unsigned char *>(frameCache.at(j).payload().constData());
            dataLen = frameCache.at(j).payload().length();

            byteGraphX.append(j);
            for (int bytcnt = 0; bytcnt < dataLen; bytcnt++)
            {
                if (byteGraphY.size() <= bytcnt) {
                    byteGraphY.append(QVector<float>()); // Changed to float
                }
//////////////////////////////////////////Need some help here /////////////////////////////////////////////
                // Combine bytes and handle signed/unsigned values based on expected data format
                int combinedValue = static_cast<int>(data[bytcnt]);
                if (bytcnt < dataLen - 1) {
                    // Example for 16-bit value (2 bytes)
                    combinedValue = (data[bytcnt] << 8) | data[bytcnt + 1];
                    if (combinedValue > 32767) { // If interpreting as signed 16-bit value
                        combinedValue -= 65536;
                    }
                }
                 byteGraphY[bytcnt].append(static_cast<float>((combinedValue)));
               // byteGraphY[bytcnt].append(static_cast<float>((combinedValue)/502.27)); 
////////////////////////////////////////////////////////////////////////////////////////////////////////////////

         // i can get proper graphs  by adjusting the scaling factor. but i have to do it for all ids and payloads. 
[image]

Puneeth-kmp, 2024-09-16:
ahh its started working perfectly .. code change   byteGraphY[0].append(static_cast<int16_t>((data[5] << 8) | data[4]) / 10.0)
[image]

cproo12, 2025-04-15:
anybody aware of a fork with functional graph support? having an odd issue where values are represented perfectly fine in the signal viewer but the graph view is all out of whack.

[image]