collin80/SavvyCAN #844: Incrrect playback time, shifted by 1 row

michcior2, 2024-09-12:
When player is playing log file with "Use original frame timing from captured file", it incorrectly uses timestamps. Takes 1 row ahead. 
See screenshot. Left is player, left is the player, right is just log load. Same CSV file. No meter what is in the file. It is always. 
Linux, not verified on windows/mac
 
[image]

michcior2, 2024-09-12:
`void FramePlaybackObject::timerTriggered()
{
    sendingBuffer.clear();

    if (useOrigTiming)
    {
        //get elapsed microseconds since last tick (in case timer skips or is otherwise inaccurate, though there are no guarantees about elapsed timer either)
        quint64 elapsed = playbackElapsed.nsecsElapsed() / 1000;
        playbackElapsed.start();
        if (playbackForward)
        {
            playbackLastTimeStamp += elapsed;
            //qDebug() << playbackLastTimeStamp;
            //qDebug() << "El: " << elapsed;            
            while (peekPosition(true) <= playbackLastTimeStamp)
            {
                updatePosition(true);
                break;
            }
            if (peekPosition(true) == 0xFFFFFFFFFFFFFFFFull)
            {
                updatePosition(true); //this'll go to the next log (if there is one)
                if (currentSeqItem->data[currentPosition].timeStamp().microSeconds() > 1000)
                    playbackLastTimeStamp = currentSeqItem->data[currentPosition].timeStamp().microSeconds() - 1000;
                else playbackLastTimeStamp = 0;
            }
        }`

michcior2, 2024-09-12:
Another dirty fix. 
The main problem of this bug, seams to get fixed when:
Function:
quint64 FramePlaybackObject::peekPosition(bool forward)
{
    int peekCurrentPosition = currentPosition;
    if (forward)
    {
        if (peekCurrentPosition < (currentSeqItem->data.count())){
             //peekCurrentPosition++; //still in same file so keep going
        }
        else //hit the end of the current file
        {
            return 0xFFFFFFFFFFFFFFFFull;
        }
    }
    else
    {
        if (peekCurrentPosition > 0) peekCurrentPosition--;
        else //hit the beginning of the current sequence
        {
            return 0xFFFFFFFFFFFFFFFFull;
        }
    }
    CANFrame *thisFrame = &currentSeqItem->data[peekCurrentPosition];
    return thisFrame->timeStamp().microSeconds();
}

1) disable `//peekCurrentPosition++;` this fixes issue with playbeck time shifted by 1 row
2) remove "-1"  from codition: `f (peekCurrentPosition < (currentSeqItem->data.count()))` this fixes not sending the last position from the loaded log file. 

Obviously,  these are dirty fixes and, since I have no idea about overall picture, they need to be wisely re-implemented by the authors.

collin80, 2024-09-17:
Yes, I've noticed that it seems like it works one off so it's a little bit messed up. Certainly I do need to get that fixed.