collin80/SavvyCAN #847: An error occurs when using an ASC log file

haikhongtam, 2024-09-20:
I am using windows 11 64 bit. I have a log file in ASC format with a size of about 12mb. When I open the file, the software automatically exits and closes. I tried another way, selecting play back, but the application still exits. Even though I have 6GB of free RAM, I tried again with the CSV file and it opened again. There is a problem here, I hope you can fix it soon.

rusoku, 2024-09-20:
Is it possible to get this file for testing ?

GavStorie, 2024-09-25:
I've had similar issues and it seems to be linked to the auto detect filetype feature. Try selecting the correct file type when you are loading the file and see if that works. It works for me that way. If i try to load without selecting the file type it crashes every time. Not just ASC but many TRC and CSV files as well.

GerryFerdinandus, 2024-09-28:
Some issues found are:

**Issue 1**
Software crash in CANHacker file decoding.
https://github.com/collin80/SavvyCAN/blob/2836977e42c697b85c9d4d1d7f366cb79036f23c/framefileio.cpp#L1136
read memory address bigger than tokens[]
need to change to
```
-                    if (tokens[d + 3] != "")
+                    if (tokens.length() > (d + 3))
```
It is assumed that the DLC value is correct. If not, the program will crash. 
There are other code parts that use the same method in this file.
if (tokens[d + a_value] != "")
IMO they should also be updated to check for length()
https://github.com/collin80/SavvyCAN/blob/2836977e42c697b85c9d4d1d7f366cb79036f23c/framefileio.cpp#L903

**Issue 2**
Wrong valid file detection of this ASC issue file by _isCANHackerFile()_
https://github.com/collin80/SavvyCAN/blob/2836977e42c697b85c9d4d1d7f366cb79036f23c/framefileio.cpp#L1048
```
-                        int len = tokens[2].toInt();
-                        if (len > -1 && len < 9)
+                        bool isValidlen;
+                        int len = tokens[2].toInt(&isValidlen, 10);
+                        if (isValidlen && len > -1 && len < 9)    
```                
After this additional validation check, the test ASC file is ignored as a valid CANhacker file.

**Issue 3**
Issue ASC file is not detected because savvyCAN expects multiple header items that are not present in the file.
*.ASC detection code
https://github.com/collin80/SavvyCAN/blob/2836977e42c697b85c9d4d1d7f366cb79036f23c/framefileio.cpp#L1641

It's missing:
**logged
version**

This issue test file:     
```
date Sat May 18 02:09:30 PM 2024
base hex timestamps absolute

000134.714376	1	178	Rx	d	8	9E 2D 80 35 F2 80 35 FF
etc..
```

Note that the *.ASC generated by savvyCAN is also not detected because it uses the word
**logging**
and not **logged**
https://github.com/collin80/SavvyCAN/blob/2836977e42c697b85c9d4d1d7f366cb79036f23c/framefileio.cpp#L1925

*.ASC file generated by savvyCAN
```
date ma sep. 23 10:00:54.792 p.m. 2024
base hex  timestamps absolute
no internal event logging
// version 11.0.0
```
A possible solution is to detect only the first two *.ASC lines and not four lines.

GavStorie, 2025-05-06:
Does anyone know if this issue has been fixed in the latest release?