collin80/SavvyCAN #857: Frames showing in console but not in the main window

AriesBreath, 2024-11-01:
Hello, I'm trying to read the CAN frames of my motorcycle.
I'm on Windows 10 and I'm using and Arduino Nano connected to a MCP2515 to connect to my diagnostic port. On the Arduino IDE I can correctly read the frames. When I start SavvyCAN I make a new connection with the LAWICEL protocol, set speed at 250000 and then enable console, from which I can see frames coming in, but after I close the connection window nothing happens on the main window and no frames are shown. I tried multiple binaries and the last development version but I can't seem to get frames.

DNThomas, 2024-11-04:
The data in your Device Console are not standard LAWICEL formatted serial frames.

In the LAWICEL format, your frames should start with either "54" (T) or "74" (t) denoting an extended or standard CAN frame.

Oddly, I came across this issue because I am having a similar problem with SavvyCAN v213 not showing extended frames 👍 
[image]

As you can see my frames are T18DBEFF13023E00 and *should* be displayed as ID=18DBEFF1 and a 2 byte packet of 3E00.

DNThomas, 2024-11-04:
... and slap me for being daft .. tried with the latest QT6 dev build and it shows them fine :)

[image]

AriesBreath, 2024-11-05:
So it's a problem with how the data is sent to the serial? I already found a solution by using another software but can you send here the exact format the frames should have? Just so if someone else has the same problem can see the solution

DNThomas, 2024-11-05:
Sorry for not including the link. Here is the manual for the original Lawicel CANUSB dongle.

[https://www.canusb.com/files/canusb_manual.pdf](url)