collin80/SavvyCAN #861: SavvyCAN，How to package running software for windows?

puwenjie, 2024-11-05:
In windows environment, I use MinGW's **windeployqt** to package the project, but why the directory of the packaged files is not the same as the SavvyCAN-Windows_x64_CIBuild directory that you packaged, I would like it to be exactly the same as yours, may I ask how you packaged windows?
I also read your readme which is about linux packaging process, Can you provide me with the details of your packaging process for Windows programs? 
hope to get your answer, thanks!
your SavvyCAN-Windows_x64_CIBuild ：
[image]
my:
[image]

collin80, 2024-11-05:
The CI build uses Visual Studio command line tools, not MinGW. Also, you can easily see all the steps by looking at the CI build script here:

https://github.com/collin80/SavvyCAN/blob/master/.github/workflows/build.yml

puwenjie, 2024-11-06:
Thanks for your answer！
I noticed the build.xml config file earlier. How can I integrate this build.xml config file and the CI build method into QT Creator so that I can debug online, does CI build support integration into QT Creator?
How did you do it?

puwenjie, 2024-11-06:
Is the CI Build you are referring to msvc? I as well configured it, in qt Creator, but the program generated by release is not the same structure as the one you packaged, why?