collin80/SavvyCAN #864: Black background text readable only by selection

belliriccardo, 2024-11-08:
Hi! My OS is Windows 10, and with the latest Qt6 [continuous release build](https://github.com/collin80/SavvyCAN/releases/tag/continuous-QT6) (this build in particular: [https://github.com/collin80/SavvyCAN/actions/runs/11735256623](https://github.com/collin80/SavvyCAN/actions/runs/11735256623)), I get the exact same behavior as #427 : black screen when "interpret frames" is selected, and text visible only by highlighting. Same as the issue above describes it. See attachment:

[image]

Strangely enough this doesn't happen to me with the [V213](https://github.com/collin80/SavvyCAN/releases/tag/V213) release (same dbc, same can configuration, same pc, same adapter, etc), which uses Qt5.
I opened a new issue because at first I thought that it was just a Qt6 problem but seeing that it existed in previous versions as well  (build from issue #427 is a Qt5 build) makes me believe that that is not the case, and maybe is worth of a separate issue.

guineawheek, 2024-11-14:
can recreate on mac os 14

guineawheek, 2024-11-14:
Update: if you select "Ignore DBC message colors" in Preferences it renders fine, but the default is unusable

collin80, 2024-11-20:
Yes, it's a bug. Sometimes very odd colors get set in the DBC file (almost certainly my fault) and then the text will render black on black or gray on black. Either way, it sucks. The solution is to check the "Ignore DBC message colors" checkbox in preferences as mentioned above. Then reasonable colors are automatically used for all signals and things look right. I need to fix this problem so I've tagged it as a bug so hopefully I remember to get this fixed.