collin80/SavvyCAN #869: How can I change the timestamps for the x-axis in graph to just a numeric quantity starting at 0?

puwenjie, 2024-11-24:
How can I change the timestamps for the x-axis in graph to just a numeric quantity starting at 0? Now that I've read it's possible to adjust it to a format like seconds, milliseconds, microseconds, etc.
Looking forward to your reply.

jonnyspizza, 2024-12-04:
Have you tried clicking the "Normalise Frame Timing" button? I think this does what you are trying to do (sets first frame to zero and then other frames are an offset from that)

puwenjie, 2024-12-20:
> Have you tried clicking the "Normalise Frame Timing" button? I think this does what you are trying to do (sets first frame to zero and then other frames are an offset from that)

Thank you for your answer, I tested and found that this item seems to just convert the data to start from moment 0, the data type is still time, not a numeric type.
Is there another way to do this?

jonnyspizza, 2025-02-27:
Not that I'm aware of sorry