collin80/SavvyCAN #886: DBC Editor fails to reopen Signal Editor for a Signal that has its name edited.

mehmetalianil, 2025-01-22:
Steps: 

Open SavvyCAN
Open the DBC File Manager
Create a new DBC File
On the default node, create a message
And in that message create a signal
Double click on that signal to open its editor
**Rename the Signal** 
Close the editor
Double click on that signal again
The editor does not appear again, the signal is not anymore editable. 

Logs: 
I couldn't find a way to come up with a trace or something (please direct me if so)
But I ran the binary through terminal an these were the STDOUT: 
```
Loaded DBC file "<Unsaved File>"  (bus: -1 , Matching Criteria: 0 Filter labeling:  enabled )
QVariant::load: unknown user type with name QList<int>.
```

Machine: 
MacBook Air M2 2023
Mac OS Sonoma 14.2

cproo12, 2025-04-11:
did you ever get around this or figure it out? I find it hard to believe people are using and developing this software without basic functionality such as this

mehmetalianil, 2025-04-11:
> did you ever get around this or figure it out? I find it hard to believe people are using and developing this software without basic functionality such as this

I would try version 213.  Some of the issues were resolved with that.

esaulenka, 2025-06-21:
it's already fixed in the latest release (V220)