collin80/SavvyCAN #910: Switching Speed and Pressing save bus settings crashes application

devilmastah, 2025-04-06:
I am running into an issue and hope to get some pointers. 
I build SavvyCan (latest code) on an RPI5, build fine and starts up fine. 

In the connection settings i get my ESP32RET to show up, like it did on my windows setup (so i know the ret is fine, it works on my laptop).
(its a Esp32Wroom with 1 canbus converter connected which works fine in windows so far)
For debugging / logging purposes i installed a RPI5 in a car i want to reverse engineer. 

Steps i do till the error: 

> Open connection window
> Add new device connection
> Serial connection (GVRet)
> Select port (ttyUSB0)
> Create new connection

The new connection shows up in the list and after 2 seconds shows Buses 2 Connected. Subtype gvret. 

When i enable the console i get data back. 
I then select 250000 as a speed enable bus and save bus settings. 
> exit out of the connection settings window, nothing shows in the can messages. 
> go back to connection settings, speed is back to 500000
Now the critical part; 
when i set the speed back to 250000 and confirm by pressing save bus settings the application hangs for a second then exits. 
In the console i am left with this: 

admin@car-pc:~/savvyCAN $ SavvyCAN 
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
qt.core.qobject.connect: QObject::connect: No such signal QSerialPort::error(QSerialPort::SerialPortError)
Segmentation fault
admin@car-pc:~/savvyCAN $ 

Any suggestions would be really welcome. I already tried rebuilding SavvyCan with 250 as default everywhere. But that does not seem to help either. 
(I guess it takes the default speed from the ESP32).

DK-custom, 2026-10-10:
I can reproduce a related GUI deadlock on current `master` (`2f683fd`) when
connecting, disconnecting, resetting, or saving bus settings for a threaded
serial backend.

The common cause appears to be synchronous cross-thread calls in
`CANConnection`:

- `CANConnection::getBusSettings()` uses `Qt::BlockingQueuedConnection`.
- `CANConnection::setBusSettings()` uses `Qt::BlockingQueuedConnection`.
- `CANConnection::suspend()` uses `Qt::BlockingQueuedConnection`.

The GUI thread waits indefinitely when the connection worker is busy in serial
processing, has not entered its event loop yet, or has already stopped. This
also explains why the problem can appear during Reset or immediately after a
connection failure rather than only while saving settings.

All current backend implementations of `piGetBusSettings()` only return the
cached `mBusData`, so a blocking hop to the worker thread is unnecessary for a
read. I tested the following minimal behavior change:

- Read the cached settings through `getBusConfig()`.
- Queue `setBusSettings()` and `suspend()` with `Qt::QueuedConnection`.
- Do not make the Connections window wait for serial-backend completion.

With Qt 5.15.2 on Windows x64 and a LAWICEL/SLCAN adapter on COM5, repeated
Connect, Disconnect, Reset, and Save Bus Settings operations no longer freeze
the UI.

One compatibility consideration: after changing `setBusSettings()` to a queued
call, callers must not assume that hardware reconfiguration has completed when
the method returns.

Relevant code:

- `connections/canconnection.cpp`: `getBusSettings`, `setBusSettings`, `suspend`
- `connections/connectionwindow.cpp`: connection control paths

Tom-evnut, 2026-10-10:
I hit this on the Qt 6 CI build (`continuous-QT6`, `QT6WIP` @ `8675061`) on Windows 10 with an ESP32RET device over GVRET serial. Changing the speed and pressing **Save bus settings** crashes every time, with an access violation (`c0000005`) inside SavvyCAN.exe at the same offset each time.

**Cause on Qt 6:** `GVRetSerial::piSetBusSettings()` builds the `0x05` (setup CAN bus) and `0x0E` (setup extended buses) commands by writing indexes into an empty `QByteArray`:

```cpp
QByteArray buffer;
buffer[0] = (char)0xF1;
buffer[1] = 5;
...
```

In Qt 5, `operator[]` returned a `QByteRef` that grew the array on these writes. In Qt 6, `operator[]` on an empty array returns a reference into the shared, read-only empty buffer, so the first write faults. `piSendFrame()` was already converted to `append()`, which is why sending frames works and only saving bus settings crashes.

**Fix:** build both commands with `append()`:
https://github.com/Tom-evnut/SavvyCAN/commit/a5d7ac05601f742865281abddbc6a68f23e23459

Tested on hardware with a CI build of that branch: speed changes now save without a crash, and traffic continues at the new speed.

The original report is a Qt 5 build on a Raspberry Pi, so it may also involve the blocking cross-thread calls @DK-custom described above. This fix covers the Qt 6 crash. I'm happy to open a PR against `QT6WIP` if that's useful.