collin80/SavvyCAN #916: custom send window

saravanan-1988, 2025-04-18:
Hi ,

I have explored the SavvyCAN for my project, it looks good.

Am trying to use **custom** option under send frames but there is no messages sent from SavvyCAN.

At a same time if i use another window under trace window, its sending CAN frames as expected.

The same i tried with scripting window, its working as expected.

Version:V213.

Please help me to resolve.

Also another request how do i implement CRC and RC messages? using SaVVYCAN?

saravanan-1988, 2025-04-25:
@collin80 can you please help me on the above issue

saravanan-1988, 2025-04-25:
Hi am using VN1610 and also tried with VirtualCAN both it was not working

saravanan-1988, 2025-04-25:
Windows

saravanan-1988, 2025-04-28:
@rusoku2 Can you please share if u have any updates

rusoku2, 2025-04-29:
> [@rusoku2](https://github.com/rusoku2) Can you please share if u have any updates

I will check.

saravanan-1988, 2025-04-29:
@rusoku2   Also another request, is there any example java script available for CRC and RC calculation? 

My idea is just to modify java script based on messages in my DBC and use send it CRC and RC related messages.

rusoku2, 2025-05-04:
It seems like everything is working.
I recompiled it with the latest version from github and adapted it to the QT v6.9.0 framework.

- git commit 75892dd
- QT framework v6.9.0
- plugin QT SerialBus for RUSOKU TouCAN adapters
- hardware RUSOKU TouCAN adapter 1x

I'm considering adding the CRC calculation now.

[image]

saravanan-1988, 2025-05-05:
Hi @collin80 Thank you for the support, I am still trying to send but not able to achieve this. can you please look at this video

https://github.com/user-attachments/assets/586153e4-8e6d-480c-a6f6-ea2f236e6fd2

gedass2000, 2025-05-05:
IMHO 100MS should be in the Trigger field

saravanan-1988, 2025-05-05:
Sorry @gedass2000 tried that option too, please find the video

https://github.com/user-attachments/assets/b019f176-e2c4-4e64-9298-721f0d3f2ce0

rusoku2, 2025-05-05:
https://github.com/user-attachments/assets/01392be8-575d-4252-b143-0b6eaa9add24

saravanan-1988, 2025-05-07:
@rusoku2 in custom window, how to utilize trigger option, i could see the Message ID, Signal and Value, Delay.. Options

It looks great value add for manual configuration and save grid use it validation.

But somehow its not working, kindly help me to resolve the same.

[image]

chriskinal, 2025-05-29:
I am seeing this same issue with MaOC Sequoia, CANAble Dongle. The Frame Sender I can load and save grids but it will not send packets. The window on the bottom of the main window that looks like "Frame Sender" will send packets but I can't load or save that grid.
[image]

minoseigenheer, 2025-11-04:
Is there any progress on this? Sending custom frames works from the lower section on the main window but not from the custom frame sender window where a grid can be saved and loaded.