collin80/SavvyCAN #922: QT6 Build Crash on Custom Send SLCAN

garnerm91, 2025-05-03:
Using SLCAN on the development build in Windows 11 and on Windows 10 crashes and exits without an error message, but receiving works fine. The device connected is a CANable V1, and it was also tested with a CANable Pro V1. The previous release build works. Below is an example of something that will make it crash. This same frame would send on the old build. 

[image]

If there is anymore info that would help resolve the issue let me know.

rusoku2, 2025-05-04:
It seems like everything is working.
I recompiled it with the latest version from github and adapted it to the QT v6.9.0 framework.

- git commit 75892dd
- QT framework v6.9.0
- plugin QT SerialBus for RUSOKU TouCAN adapters
- hardware RUSOKU TouCAN adapter 1x
- Windows and MacOS versions

[image]

garnerm91, 2025-05-06:
> It seems like everything is working. I recompiled it with the latest version from github and adapted it to the QT v6.9.0 framework.
>
> * git commit [...]

I haven't been able to get the main branch to compile.

Both of my Canables, rev A and PRO v1, have the same performance of SavvyCAN 214 crashing. You have V213 compiled per your screenshot.

garnerm91, 2025-05-07:
Resolved it. Windows was killing the process due to a buffer overflow, I believe. Resizing the buffer fixed it for me. 

`bool LAWICELSerial::piSendFrame(const CANFrame& frame)
{
    QByteArray buffer;
    int c;
    quint32 ID;

    qDebug() << "Sending out lawicel frame " << " on bus " << frame.bus;

    framesRapid++;

    if (serial == nullptr) return false;
    if (serial && !serial->isOpen()) return false;
    //if (!isConnected) return false;

    // Doesn't make sense to send an error frame
    // to an adapter
    if (frame.frameId() & 0x20000000) {
        return true;
    }

    ID = frame.frameId();
    if (frame.hasExtendedFrameFormat()) ID |= 1u << 31;

    qDebug() << "ID " << ID;

    int idx = 0;
    QString buildStr;
    if(frame.hasFlexibleDataRateFormat()){
        if (frame.hasExtendedFrameFormat())
        {
            if (frame.hasBitrateSwitch())
                buildStr = QString::asprintf("B%08X%u", ID, LAWICELSerial::bytes_to_dlc_code(frame.payload().length()));
            else
                buildStr = QString::asprintf("D%08X%u", ID, LAWICELSerial::bytes_to_dlc_code(frame.payload().length()));
        }
        else
        {
            if (frame.hasBitrateSwitch())
                buildStr = QString::asprintf("b%03X%u", ID, LAWICELSerial::bytes_to_dlc_code(frame.payload().length()));
            else
                buildStr = QString::asprintf("d%03X%u", ID, LAWICELSerial::bytes_to_dlc_code(frame.payload().length()));
        }
    }
    else {
        if (frame.hasExtendedFrameFormat())
        {
            buildStr = QString::asprintf("T%08X%u", ID, static_cast<unsigned int>(frame.payload().length()));
        }
        else
        {
            buildStr = QString::asprintf("t%03X%u", ID, static_cast<unsigned int>(frame.payload().length()));
        }
    }
    int neededSize = buildStr.length() + (frame.payload().length() * 2) + 1;  // +1 for CR
    buffer.resize(neededSize);

    foreach (QChar chr, buildStr)
    {
        buffer[idx] = chr.toLatin1();
        idx++;
    }

    for (c = 0; c < frame.payload().length(); c++)
    {
        QString byt = Utility::formatByteAsHex(frame.payload()[c]);
        buffer[idx + (c * 2)] = byt[0].toLatin1();
        buffer[idx + (c * 2) + 1] = byt[1].toLatin1();
    }
    buffer[idx + (frame.payload().length() * 2)] = 13; //CR

    sendToSerial(buffer);

    return true;
}`

I added 
`    int neededSize = buildStr.length() + (frame.payload().length() * 2) + 1;  // +1 for CR
    buffer.resize(neededSize);`

closed 2025-05-07 by garnerm91

gedass2000, 2025-05-07:
There are no problems with the latest release either. But I see that you found a bug in the SLCAN interface. TouCAN uses the QtSerialBus native plugin, which is probably why the program behaves differently.
There is ping pong between two adapters.

[image]

garnerm91, 2025-05-07:
> There are no problems with the latest release either. But I see that you found a bug in the SLCAN interface. TouCAN uses the QtSerialBus native plugin, which is probably why the program behaves differently. There is ping pong between two adapters.
> 
> [image]

I see I didn't catch that they were using a different interface. 

Resizing the buffer fixes this issue though