collin80/SavvyCAN #925: Unable to type scripts in scripting interface

italocjs, 2025-05-06:
Writing, copying and pasting from the tool no longer works it.
Using compiled version from current master branch.

Commands as CTRL+A, CTRL+C, CTRL+V are not working either.

[image]

Below is the app log from terminal: https://pastebin.com/YuCu8C5b

I noticed many 
```shell
** (SavvyCAN:203540): WARNING **: 10:52:43.580: AT-SPI: Could not obtain desktop path or name
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()

```

italocjs, 2025-05-06:
It works if i start savvy as `QT_QPA_PLATFORM=xcb ./SavvyCAN`

Here are system details:

```shell
italo@italo-Vostro-3520:~/SavvyCAN$ cat /etc/os-release
PRETTY_NAME="Ubuntu 24.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.2 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo
italo@italo-Vostro-3520:~/SavvyCAN$ uname -r
6.11.0-25-generic
italo@italo-Vostro-3520:~/SavvyCAN$ echo $XDG_SESSION_TYPE
wayland
italo@italo-Vostro-3520:~/SavvyCAN$ qmake -v
QMake version 3.1
Using Qt version 5.15.13 in /usr/lib/x86_64-linux-gnu
italo@italo-Vostro-3520:~/SavvyCAN$ uname -m
x86_64
italo@italo-Vostro-3520:~/SavvyCAN$ 

```

raiderss, 2025-05-09:
The documentation in this repository is excellent! Very clear and comprehensive.