collin80/SavvyCAN #961: Permission error when attempting to connect with EVTV ESP32-WROOM-32 board.

wombatus-1, 2025-07-20:
I've set up up savvyCAN on Ubuntu 2 without 4-04-01 (after resolving problems with FUSE).  saavyCAN opens OK without any connections.  I then connected a newly delivered EVTV ESP32-WROOM-32 board. The connection seemed to be recognised as ttyUSB0, but nothing at all happened until I did a reset (from saavyCAN).  The reset was announced in the console, but indicated a "permission error".  I'm not sure this problem is an EVTV-ESP32 board issue, a saavyCAN software issue, or an Ubuntu issue.  Any reasonable advice will be welcomed, as I haven't been able to find a way around the problem.

esaulenka, 2025-07-20:
Did you tried to run Savvy as root user?
Or maybe look on this receipt: https://askubuntu.com/questions/133235/how-do-i-allow-non-root-access-to-ttyusb0

wombatus-1, 2025-07-21:
> esaulenka

Just in case my message didn't get through on the forum, I must thank you!  Running savvyCAN as root user worked fine.

Regards.

Clive.

Dr Clive Boughton

97 Bankers Road
Bywong, NSW 2621
Ph: 0410 632 055

esaulenka, 2025-07-21:
You can run (only once) in the terminal
`sudo usermod -a -G dialout $USER`
and reboot your PC.
This will provide access to the 'modem' (all serial ports still considered as modems in modern Linux) for your unpriveleged user account.
And I agree, it may be stated somwhere in the documentation.